udacity/robotics-beta-test-issues #310: L05E02 - Rebuild workspace after creating python script

alvarovillena, 2017-05-16:
I think there is no need to rebuild the workspace after creating a new script. When you add a python script, you don't need to build anything new.

Shreeyak, 2017-05-16:
What about c++?

alvarovillena, 2017-05-16:
If you add c++ code you need to build the package, of course. But in this lesson you are asked to create just a python script.
Maybe the rebuild step is there in case you download again the workspace instead of using the one created in the previous lessons.

Shreeyak, 2017-05-16:
And also as a good habit. :)

bkinman, 2017-05-19:
Until students understand when they do and when they don't need to rebuild the workspace, I'd prefer to have them rebuild. For example, python scripts in the `src` directory, which are installed as modules, require the workspace to be built at least once to be used.

I don't want to bog students down with this depth of detail here.

I think that not remembering to source your workspace, and not remembering to build your workspace will be common problem areas that students run into. I'd like to avoid this.

closed 2017-05-19 by bkinman