udacity/robotics-beta-test-issues #312: P1 Sim

jchernus, 2017-05-16:
Sim crashed in manual mode when I try to pick up a rock. 

Just after I pick up my first rock in auto mode, the communication that I see in my terminal speeds up threefold, and continues for 5 seconds after I shutdown the sim (as if there was a whole bunch more buffered). 

Seems like a sim-related problem.

ryan-keenan, 2017-05-21:
Hi Julia, I haven't seen this problem myself.  Are you running at high resolution / graphics quality?  any change in the problem if you reduce?  What kind of machine are you running on?

jchernus, 2017-05-22:
Ryan, turns out this was being caused by me sending the "pick up" command more than once. Not an issue after I fixed that problem in my code. Couldn't figure it out for while, sorry!

jchernus, 2017-05-22:
Manual mode no longer crashes either. Don't know what changed there...

ryan-keenan, 2017-05-22:
Great!  I'll close this now

closed 2017-05-22 by ryan-keenan