udacity/robotics-beta-test-issues #313: all terminal commands should be in text to copy/paste

867sec, 2017-05-16:
all terminal commands should be in text to copy/paste
in Lesson 3 Part 4 - Environment setup the following is a screenshot but should be in text for easy copy/paste: "source /opt/ros/kinetic/setup.bash"

bkinman, 2017-05-19:
We think it's probably helpful to have students typing commands as long as it's the kind of stuff they should be expected to remember. We think making them type out long cumbersome commands for installation processes or other stuff they'll probably be looking up is less useful. 

This stance could change if we continue to get strong feedback though.

closed 2017-05-19 by bkinman

867sec, 2017-05-19:
Typing installation process commands is a recipe for disaster as the
students can mess up environment that would be very hard to troubleshoot by
mentor / reviewers. Commands that are easy to copy paste are a great
solution... I never learned by heart commands that some tutorial made me
type once... You learn commands by heart when you need to use them 10 times
a day in your everyday work... Please make them copy/paste possible.