udacity/robotics-beta-test-issues #315: duncanchan L3E3

duncanchan, 2017-05-16:
It is possible to have a VHD for Windows hyper-V as well? For windows user, we may not need VMWare, and thus no need to worry about trial expiration.

duncanchan, 2017-05-16:
Also, the instructions to use vmdk from the video: https://kb.vmware.com/selfservice/microsites/search.do?language=en_US&cmd=displayKC&externalId=2010196 is outdated. The VMWare tool has a different UI now. Maybe we should have a github page outline the installation step more clearly.

bkinman, 2017-05-19:
In windows VMWare Player is free, and should not expire.... This being said, I would like to provide a Hyper-V image at some point; especially if VMWware does not work for some students.

I agree that our instructions need to be more up to date. I'll go ahead and leave this issue open until we update the instructions. Thanks!

bkinman, 2017-05-23:
We are in the process of creating a github repo with instructions and FAQ for VM. Closing this.

closed 2017-05-23 by bkinman