udacity/robotics-beta-test-issues #320: duncanchan L05E02

duncanchan, 2017-05-16:
This lesson has a lot of links to older lessons. I think it suffices to use notation L05E02, rather than a link that opens the classroom in a new tab. It's hard to track which classroom is which.

duncanchan, 2017-05-16:
maybe also explain what chmod +x means.

bkinman, 2017-05-18:
Hrmm. We should be able to use both links and some notation, I think.
Unfortunately, we are not 100% solid on the notation at the moment, as this lesson and concept numbering is automatically assigned.

I'm going to go ahead and leave this open for now, and I'll fix it when things become a little more solid.

bkinman, 2017-05-23:
Closing this due to lack of context. Please provide more context (such as the title of the section) when submitting an issue, as the lesson and concept numbers have changed a great deal, and will continue to change.

closed 2017-05-23 by bkinman