udacity/robotics-beta-test-issues #331: arek - error running roslaunch simple_arm robot:

867sec, 2017-05-16:


bkinman, 2017-05-17:
This is a known issue. Sadly, ever since I've begun using gazebo, I've had problems with it crashing when starting up.

I'm going to start up a thread of conversation with folks at OSRF to see if they can fix this problem, as it will certainly be causing grief for students.

With what frequency do you see this error?

bkinman, 2017-05-17:
If I don't get a timely response from OSRF, I'll go ahead and add a provision somewhere mentioning this problem.

867sec, 2017-05-17:
The error happens intermittently. Hard to duplicate.
What does OSRF stand for?

bkinman, 2017-05-18:
Open Source Robotics Foundation, the maintainers of ROS/Gazebo.

Shreeyak, 2017-05-21:
@bkinman So is there any information so far about anything that might be causing the crash?

luqiang21, 2017-05-21:
I have this same error every time I run
  rosrun simple_arm simple_mover
I have no error when I only run 
  roslaunch simple_arm robot_spawn.launch

bkinman, 2017-05-22:
luqiang21: How many times did you try? And how long did you wait before retrying?

luqiang21, 2017-05-23:
@bkinman I tried about 5 times. Every time I run: rosrun simple_arm simple_mover, the gazebo was opened and then stopped, then errors shown in terminal. And I have updated gazebo to 7.7.0, increased the VM memory to 2G. I tried one more time just now, below are all the information I have:

robo@robo-virtual-machine:~/catkin_ws$ roslaunch simple_arm robot_spawn.launch 
... logging to /home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/roslaunch-robo-virtual-machine-10980.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robo-virtual-machine:43659/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /simple_arm/joint_1_position_controller/joint: joint_1
 * /simple_arm/joint_1_position_controller/pid/d: 10.0
 * /simple_arm/joint_1_position_controller/pid/i: 0.01
 * /simple_arm/joint_1_position_controller/pid/p: 100.0
 * /simple_arm/joint_1_position_controller/type: effort_controller...
 * /simple_arm/joint_2_position_controller/joint: joint_2
 * /simple_arm/joint_2_position_controller/pid/d: 10.0
 * /simple_arm/joint_2_position_controller/pid/i: 0.01
 * /simple_arm/joint_2_position_controller/pid/p: 100.0
 * /simple_arm/joint_2_position_controller/type: effort_controller...
 * /simple_arm/joint_state_controller/publish_rate: 50
 * /simple_arm/joint_state_controller/type: joint_state_contr...
 * /use_sim_time: True

NODES
  /simple_arm/
    spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [10994]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 32a3f5b2-3f59-11e7-b95a-000c294eadff
process[rosout-1]: started with pid [11007]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [11010]
process[robot_state_publisher-3]: started with pid [11014]
process[simple_arm/spawner-4]: started with pid [11021]
process[gazebo-5]: started with pid [11031]
process[gazebo_gui-6]: started with pid [11039]
process[urdf_spawner-7]: started with pid [11044]
SpawnModel script started
[ INFO] [1495503806.311152725]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1495503806.332202756]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1495503806.531325, 0.000000]: Loading model XML from ros parameter
[INFO] [1495503806.566333, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1495503807.953119, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1495503808.744431, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1495503810.038996, 261.179000]: Spawn status: SpawnModel: Model pushed to spawn queue, but spawn service timed out waiting for model to appear in simulation under the name simple_arm
[urdf_spawner-7] process has finished cleanly
log file: /home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/urdf_spawner-7*.log
[ INFO] [1495503812.463726227, 261.179000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1495503812.546170955, 261.179000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1495503812.755044836, 261.179000000]: Loading gazebo_ros_control plugin
[ INFO] [1495503812.762254004, 261.179000000]: Starting gazebo_ros_control plugin in namespace: /simple_arm
[ INFO] [1495503812.788553110, 261.179000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[... 4 lines cut ...]
[INFO] [1495503813.808010, 261.179000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1495503813.831854, 261.181000]: Loading controller: joint_state_controller
[ INFO] [1495503813.960691358, 261.199000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1495503814.417144936, 261.231000000]: Physics dynamic reconfigure ready.
[INFO] [1495503815.789541, 261.411000]: Loading controller: joint_1_position_controller
gzserver: symbol lookup error: /opt/ros/kinetic/lib//libeffort_controllers.so: undefined symbol: _ZN4urdf5Model23initParamWithNodeHandleERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN3ros10NodeHandleE
[gazebo-5] process has died [pid 11031, exit code 127, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/robo/catkin_ws/src/simple_arm/worlds/willow_garage.world __name:=gazebo __log:=/home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/gazebo-5.log].
log file: /home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/gazebo-5*.log
close failed in file object destructor:
sys.excepthook is missing
lost sys.stderr
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 190, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
[... cut ...]
[WARN] [1495503849.540742, 261.762000]: Controller Spawner couldn't reach controller_manager to take down controllers. Waited for 30 second(s).
[simple_arm/spawner-4] process has died [pid 11021, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner joint_state_controller joint_1_position_controller joint_2_position_controller __name:=spawner __log:=/home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/simple_arm-spawner-4.log].
log file: /home/robo/.ros/log/32a3f5b2-3f59-11e7-b95a-000c294eadff/simple_arm-spawner-4*.log

cornzyblack, 2017-05-23:
also having an error when i try roslaunch

robo@robo-virtual-machine:~/catkin_ws$ source devel/setup.bash
robo@robo-virtual-machine:~/catkin_ws$ roslaunch simple_arm robot_spawn.launch
... logging to /home/robo/.ros/log/f15982f2-3fc1-11e7-989f-000c29ed56e5/roslaunch-robo-virtual-machine-5815.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robo-virtual-machine:43811/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /simple_arm/joint_1_position_controller/joint: joint_1
 * /simple_arm/joint_1_position_controller/pid/d: 10.0
 * /simple_arm/joint_1_position_controller/pid/i: 0.01
 * /simple_arm/joint_1_position_controller/pid/p: 100.0
 * /simple_arm/joint_1_position_controller/type: effort_controller...
 * /simple_arm/joint_2_position_controller/joint: joint_2
 * /simple_arm/joint_2_position_controller/pid/d: 10.0
 * /simple_arm/joint_2_position_controller/pid/i: 0.01
 * /simple_arm/joint_2_position_controller/pid/p: 100.0
 * /simple_arm/joint_2_position_controller/type: effort_controller...
 * /simple_arm/joint_state_controller/publish_rate: 50
 * /simple_arm/joint_state_controller/type: joint_state_contr...
 * /use_sim_time: True

NODES
  /simple_arm/
    spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [5829]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f15982f2-3fc1-11e7-989f-000c29ed56e5
process[rosout-1]: started with pid [5842]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [5845]
process[robot_state_publisher-3]: started with pid [5851]
process[simple_arm/spawner-4]: started with pid [5862]
process[gazebo-5]: started with pid [5869]
process[gazebo_gui-6]: started with pid [5879]
process[urdf_spawner-7]: started with pid [5887]
SpawnModel script started
[INFO] [1495548795.967055, 0.000000]: Loading model XML from ros parameter
[INFO] [1495548795.993482, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[gazebo-5] process has died [pid 5869, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/robo/catkin_ws/src/simple_arm/worlds/willow_garage.world __name:=gazebo __log:=/home/robo/.ros/log/f15982f2-3fc1-11e7-989f-000c29ed56e5/gazebo-5.log].
log file: /home/robo/.ros/log/f15982f2-3fc1-11e7-989f-000c29ed56e5/gazebo-5*.log
[INFO] [1495548797.648343, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[WARN] [1495548827.687668, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[simple_arm/spawner-4] process has finished cleanly
log file: /home/robo/.ros/log/f15982f2-3fc1-11e7-989f-000c29ed56e5/simple_arm-spawner-4*.log

bkinman, 2017-05-23:
@luqiang21 How long did you wait before retrying? This was a significant question. Can you try waiting 30 seconds after killing before relaunching and get back to me?

@cornzyblack Can you do the same?

luqiang21, 2017-05-23:
@bkinman I waited for about 1 minute. I was prompted that the gazebo is not working properly do I want to restart? I selected yes, but nothing happens. Then I restarted the command, same results.

hvpandya, 2017-05-24:
@bkinman some of these error say 
`Controller Spawner couldn't find the expected controller_manager ROS interface.`

Don't think those are gazebo issues

bkinman, 2017-05-24:
It seems like we have a couple of different errors happening here guys.

> `Controller Spawner couldn't find the expected controller_manager ROS interface.`
This is a warning, and is the expected output before `rosdep -i has been run.` (i.e. prior to [this concept](https://classroom.udacity.com/nanodegrees/nd209/parts/af07ae99-7d69-4b45-ab98-3fde8b576a16/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/658c94f5-f806-4273-9001-9e2838e56856/concepts/ad9785ee-a561-4a98-bf11-a8ef32d6fc75)). 

I am also seeing failures due to gazebo segfault, which is the issue at topic here.

Lastly, I am seeing an "undefined symbols error"... This error is likely due to a mismatch in versions of installed ros packages... I'll bet that an apt-get upgrade would fix this.... But yet again, this is a separate issue. Let's try to separate our concerns here.

bkinman, 2017-05-24:
@cornzyblack: Can you create a separate issue? I have a slew of questions I'd like to ask you about that one.

cornzyblack, 2017-05-24:
Someone said I should do an update to gazebo 7.7. I did so and It fixed my problems. Thank you

luqiang21, 2017-05-24:
@bkinman I used the latest vmdk, finally gazebo works!

bkinman, 2017-05-24:
I'm going to go ahead and lock this issue for now, as this thread is getting pretty long, and I don't want to lose sight of the segfaulting issue. I'll keep it open in the hopes that I get an update on my OSRF ticket.

bkinman, 2017-06-30:
This is a known issue, and I'm not hearing about any movement with the OSRF issue I filed, going to put this in the backlog. There's not much we can do unless we want to dive into the gazebo source code and fix this ourselves. There _is_ a bit of a working solution in that students can wait, and then start gazebo again.

closed 2018-05-03 by jessluls