udacity/robotics-beta-test-issues #332: Pick and place - Project not working

alvarovillena, 2017-05-16:
After launching `./safe_spawner.sh`, this error appears in the `cafe.launch` terminal:

``` [ERROR] [1494958183.196602699]: Exception while loading planner 'ompl_interface/OMPLPlanner': Failed to load library /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libompl.so.12: cannot open shared object file: No such file or directory)
Available plugins: ompl_interface/OMPLPlanner ```

After that, I get warnings and other 2 errors:
```[WARN] [1494958213.151508, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.```
```[ERROR] [1494958268.850521176, 1094.674000000]: Action client not connected: kuka_arm/arm_controller/follow_joint_trajectory
```
```[ERROR] [1494958285.245579393, 1109.735000000]: Action client not connected: kuka_arm/gripper_controller/follow_joint_trajectory```
```[ WARN] [1494958285.606696977, 1110.059000000]: MoveGroup running was unable to load ompl_interface/OMPLPlanner```

As a result, the project is not working. Some of the models in RVIZ doesn't show up, and the next button doesn't do anything.
In addition, I think that the Gazebo world is wrong. The blue cylinder is on the floor, the shelf is just a box and I can't see any dropbox.

I tested this both on my local machine and the VM, and the result is the same.

hvpandya, 2017-05-16:
The reason you see blocks inside gazebo is that Gazebo is unable to find the model files (which are present in the project you downloaded)

What does your directory structure look like?
Please _**cd**_  into /kuka_arm folder and paste the results of _**pwd**_ from within that directory

alvarovillena, 2017-05-17:
```/home/robo/catkin_ws/src/kinematics_project/kuka_arm```

I had an intermediate folder before too, but I got almost the same result. The OMPLPlanner error is not showing now, but everything else is the same.

hvpandya, 2017-05-17:
Did you replace, ros_ws with catkin_ws in your case for every instruction?

alvarovillena, 2017-05-17:
Yes I did, the `$GAZEBO_MODEL_PATH` was pointing to the right folder.
But it seems I did something else wrong. I just tried to reinstall it for the third time and now is magically working :D Thanks for your help!

hvpandya, 2017-05-17:
Alrighty, glad things worked out!

closed 2017-05-17 by hvpandya