udacity/robotics-beta-test-issues #334: gazebo crashing intermittently

867sec, 2017-05-16:


867sec, 2017-05-16:
[image]

867sec, 2017-05-16:
[image]

hvpandya, 2017-05-16:
Did you perform a rosdep install?

Your setup appears to be missing effort_controllers. Please try the following:
```
$ cd ~ros_ws
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
```

867sec, 2017-05-17:
Yes you are right!
After your suggestions and per instructions I used rosdep install -i
simple_arm
and it installed effort_controllers.
Then I run your

rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y

and got the following message:

#All required rosdeps installed successfully

so everything works now! Thanks!

closed 2017-05-17 by 867sec