udacity/robotics-beta-test-issues #335: letyrodri L03E03

letyrodridc, 2017-05-16:
I couldn't install VMware. It has an strange error, and the workaround is not working: https://communities.vmware.com/thread/430034

I have a Linux machine, so I decided to install ROS. 

Some students could run into problems in the VM installation.

Could we add the instructions to install ROS on a Ubuntu? For those that have troubles with the VM or those who use Linux and just want to use directly ROS. Or is there an additional installation to do?? 

It seems very tricky installing VMware without Pro version, could Virtual Box used instead?

hvpandya, 2017-05-17:
Did you try to install VMware Fusion or VMware Player? 

For Windows and Linux, VMware Player: [Link](https://my.vmware.com/en/web/vmware/free#desktop_end_user_computing/vmware_workstation_player/12_0) should be installed

VMware Fusion, is only for Mac and other iOS devices

letyrodridc, 2017-05-17:
VMware Player, but I had the same result as the link said.

letyrodridc, 2017-05-17:
Unless some special configuration is required, install ROS for Ubuntu users is very simple. http://wiki.ros.org/kinetic/Installation/Ubuntu

bkinman, 2017-05-17:
> Could we add the instructions to install ROS on a Ubuntu?
Yup, we will be doing this, but our current official stance is that we will not be supporting Linux as a host platform (it would be a support nightmare).

bkinman, 2017-05-17:
> Virtual Box used instead?
We've found that gazebo and virtual box do not play well together, so we have elected to standardize around VMWare for now.

bkinman, 2017-05-17:
> ROS for Ubuntu users is very simple. 

It's simple to install, but it's very easy to break a system after install; once the system is broken, it can be a troubleshooting nightmare.

At this point, we don't want to discourage people running Linux on their computers, but we won't be able to support them.

bkinman, 2017-05-17:
I'm going to go ahead and close this.

BTW, from the sounds of it, it would seem that vmwares linux kernel modules are either not installed, or are not running for some reason. Did you try restarting your system after vmware was installed? I believe that is required to get the kernel modules to launch.

closed 2017-05-17 by bkinman

letyrodridc, 2017-05-20:
Hey, just FYI I made some progress! It's turned out that I need to setup my computer BIOS, turning off UFI and enabling Intel Virtualization Technology.
I haven't finish yet, but I made some progress, now I can start the VM without Operating System.
Btw, you were right! :) installing ROS turned out in a lot of packages installation, but the worse part was that I've started to get some strange errors, so there was no more solution that try to make the VM work.