udacity/robotics-beta-test-issues #348: Can't load .vmdk on win 10

luqiang21, 2017-05-17:
I downloaded VMware Workstation 12 Player for windows. And I followed the two ways alvarovillena  mentioned in his issue to load the vmdk file and bring up the VM.
The first method:
First Method:

1. File -> Open a virtual machine
2. Select the provided Ubuntu+64-bit.vmx file
3. When the VM is started for the first time, a dialog box asks us to browse for the .vmdk file. Select the provided .vmdk file.
4. The program asks if the machine has been moved or copied. Select 'I Copied It'

Then I got following error:
This virtual machine is configured for 64-bit guest operating systems. However, 64-bit operation is not possible.

This host supports Intel VT-x, but Intel VT-x is disabled.

Intel VT-x might be disabled if it has been disabled in the BIOS/firmware settings or the host has not been power-cycled since changing this setting.

(1) Verify that the BIOS/firmware settings enable Intel VT-x and disable 'trusted execution.'

(2) Power-cycle the host if either of these BIOS/firmware settings have been changed.

(3) Power-cycle the host if you have not done so since installing VMware Player.

(4) Update the host's BIOS/firmware to the latest version.

For more detailed information, see http://vmware.com/info?id=152.

Then I followed the second method:

1. File -> Create a new virtual machine
2. Select 'I will install the operating system later'.
3. Select 'Linux' version 'Ubuntu 64-bit' as guest OS.
4. Create the machine. Hardware settings can be customized.
5. Once the VM is created, select 'Edit virtual machine settings'. Remove the Hard Disk device and add a new Hard Disk device, using an existing virtual disk. Select the .vmdk as the disk file.

Then I got the same error as above.

I tried to select Ubuntu instead of Ubuntu-64bit. I didn't have the same error, but was reminded that '

Cannot connect the virtual device sata0:1 because no corresponding device is available on the host.

Do you want to try to connect this virtual device every time you power on the virtual machine?

After I chose Yes, the VMware stopped working.

luqiang21, 2017-05-17:
My computer is lenovo x1 carbon, processor: i5-6200, 8 GB memory, 64-bit.

luqiang21, 2017-05-18:
I successfully loaded the system using the same vmdk and  VMWare Fusion Player for Mac on Mac. And I tried oracle virtualbox to load the vmdk, similarly, I got a black screen with one underscore, but can't input anything.

zenetio, 2017-05-20:
I am also having the same problem. When I try to load the .vmdk file, I getn an hd0 error and the Linux rescue prompt is enabled.

sahiljuneja, 2017-06-10:
Hey @luqiang21 and @zenetio !

Could you please confirm if VT-x was enabled for you guys in BIOS? That seems to be required for hardware assisted virtualization when working on Windows + Intel + Vmware.

luqiang21, 2017-06-10:
@sahiljuneja Vt-x was not enabled.

sahiljuneja, 2017-06-12:
@luqiang21 Did enabling it help then? If you haven't yet tried that, please do and let us know!

kylesf, 2017-06-13:
There is a new posted VM, that has been tested with windows. Please visit the classroom for an updated like. If you have problems with that one please open a new waffle ticket.

closed 2017-06-13 by kylesf