udacity/robotics-beta-test-issues #358: Source the ROS Environment - Wrong test answer

alvarovillena, 2017-05-20:
The second question on this [lesson](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/ae288087-b8ef-4e2d-97ff-a85993300ac7/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/2cd33882-d29e-43e8-9ff7-10398c8b5351/concepts/b051cbda-2e1a-4faa-9490-00d739c00245#) asks the student what happens when the bash command `./`is executed. The lesson explains the following:
`when a script executed via ./ is exited, all environment variables set by it will be lost`

As you can see on this [image](http://i.imgur.com/nFtcPde.png) the first option is the one set as correct, but I think this is wrong.

sagunms, 2017-05-20:
Yeah, I was confused by this too.

bkinman, 2017-05-22:
Oops! Fixed.

closed 2017-05-22 by bkinman