udacity/robotics-beta-test-issues #359: sagun L3-8 Your Virtual Machine - Add extra warnings

sagunms, 2017-05-20:
sagun, 03.08, Virtual Machine: Add extra warnings in the chapter for happier students

In addition to the text: 
> "The only problem is that it's a rather large download (3+GB!). So, find yourself a good internet connection and download the VM disk image."

Please consider adding the some of the following points to the paragraph. 
Just my thoughts - I feel this will save student's time and avoid frustration.

* Add a warning to keep Ubuntu+64-bit.zip archive safe in case you have to start from scratch - takes hours to re-download. I had to delete the 3 GB .zip file because I was running out of space on my small SSD drive on my MacBook. Then I accepted the software update and then didn't even have space to do anything else. 

* Give an option for the students to use Virtual Box on Mac (and/or Windows and Linux). It seems to works fine. Just need to install VBoxGuestAdditions for higher resolution screen. 

* During my first attempt, the OS prompted to update software. I clicked Yes and the I could not complete the update as the there was no space remaining. Perhaps, it would be good to add link or instructions to increase the size of the virtual disk in case the space runs out before loading the VM. Second attempt, I updated software using the terminal instead. 

```
sudo apt-get update && sudo apt-get upgrade -y
sudo apt-get autoremove
sudo apt-get autoclean
sudo apt-get clean
```

End result was 43% space used instead of 100%. 

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/f491093d-8e67-4eab-8102-87bae095ca79/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/2cd33882-d29e-43e8-9ff7-10398c8b5351/concepts/0cabcedf-a31b-4f41-b85a-3e0aedbdf5e8

sagunms, 2017-05-20:
**EDIT:** Rviz crashes with a _Segmentation fault (core dumped)_ error during startup. I tried turning off _3D Acceleration_ and different `OGRE_RTT_MODE` options as suggested in the _Segfault during startup_ section of [ROS Troubleshooting docs](http://wiki.ros.org/rviz/Troubleshooting). 
Probably adding a note in the lecture would help: 
 saying that the nano-degree supports only VMWare for the class because of issues with RViz so try Virtual Box if the student is adventurous only.

kylesf, 2017-06-10:
Okay, for keeping the archive saved for back up I will add a note in the classroom. 

As of right now, we are sticking with VMware but we are working on a docker image to help with those who do not want to pay for VMware on Mac.

The software is updated in the newest VM and only apt-get clean is ran. The running of auto remove and auto clean can cause problems with ROS. 

Let me know if I am missing anything before closing this!

kylesf, 2017-06-12:
This has been taken care of and you can find the updated VM in the classroom! Thanks for the help!

closed 2017-06-12 by kylesf