udacity/robotics-beta-test-issues #362: Shreeyak L1.1E02: Too many hi-fi terms flying around

Shreeyak, 2017-05-20:
In this video, while explaining what ROS provides, there are a lot of technical terms used in quick succession without any explanation. I feel this section will go entirely over students' heads.

It'll be nice if there was on overlay on the video that takes each benefit mentioned (eg: communications) and shows a little animation explaining it in brief. And has ~1sec pause in between.

This would require only editing the video and adding a few animations.

Shreeyak, 2017-05-21:
later videos do explain many of the terms, but it will be good to have *something* here. Being an introduction video, students should be able to understand what's being spoken about.

bkinman, 2017-05-22:
In future commentary, can you provide the title of the section that you speak of?
The numbering tends to change quite a bit as we develop.

bkinman, 2017-05-22:
If you're referring to "Build Robots with ROS", I think I see what you mean. We enumerate some features, but we don't really call them out visually. I'll create a ticket for this, so that we can perhaps put it in a future edit!

Great feedback!

closed 2017-05-22 by bkinman