udacity/robotics-beta-test-issues #364: diyjac L05E06 VM rqt_image_view fails to open drm device: Permission denied.

diyjac, 2017-05-20:
After many attempts to get the VM ROS to work, I finally found that using the update provided by VMWare after opening the Udacity provided VM would install the correct version of ROS and drivers.  But the command:
```
rqt_image_view /rgb_camera/image_raw
```
Still does not work to show the camera image.  Instead it gives the following error:
```
libGL error: failed to open drm device: Permission denied
libGL error: failed to load driver: vmwgfx
```
Could be compatibility issues with the host system?  I am using Ubuntu 16.04 with a NVIDIA GPU.  The
```
rosservice call /arm_mover/safe_move '{"joint_1": 1.57, "joint_2": 1.57}'
```
command works to point the arm towards the cabinets as can be seen in the image below.

[image]

diyjac, 2017-05-20:
A video of the problem can be found here: https://files.slack.com/files-pri/T4ET9BQ3U-F5G03FGUA/download/rosvm.mp4

hvpandya, 2017-05-21:
It seems you are having permission issues with your GPU, can you confirm that you(your username) is part of the video group?

If you are not a part of the video group you can do the following:
```sh
sudo usermod -a -G video <username>
```
then reboot and check if the problem persists.

diyjac, 2017-05-22:
I added my user in the host system to the video group.  The permission issue disappeared, but now the arm no longer actuates, and the camera appears the same.  I will attempt a reinstall of the VM.
[image]

diyjac, 2017-05-22:
I have reinstalled the VM into VMWare and am still having issues with bringing up the camera using
```
rqt_image_view /rgb_camera/image_raw
```
Produces the following error
```
. . .
[INFO] [1495423481.442633, 261.754000]: Controller Spawner: Loaded controllers: joint_state_controller, joint_1_position_controller, joint_2_position_controller
[INFO] [1495423481.446969, 261.756000]: Started controllers: joint_state_controller, joint_1_position_controller, joint_2_position_controller
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
```
And kills the gazebo ros interface:
```
Aborted (core dumped)
[gazebo-5] process has died [pid 38191, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/robo/catkin_ws/src/simple_arm/worlds/willow_garage.world __name:=gazebo __log:=/home/robo/.ros/log/2fc95870-3e9e-11e7-b205-000c296298c0/gazebo-5.log].
log file: /home/robo/.ros/log/2fc95870-3e9e-11e7-b205-000c296298c0/gazebo-5*.log
```
The following is a link to the problem captured in a video session.
https://files.slack.com/files-pri/T4ET9BQ3U-F5G52QE02/download/vm-ros-rqt-depthbuffer-incompatible.mp4

diyjac, 2017-05-22:
This appears to be a known issue: http://answers.gazebosim.org/question/9395/gazebo-crashes-when-inserting-camera-in-virtual-machine-with-ubuntu-vivid-ros-jade/ since ROS jade.  It appears that the answer is to upgrade to gazebo 7.4+.

hvpandya, 2017-05-22:
This is the exact reason why we suggest in the README to upgrade your gazebo to 7.7+ version

```sh
$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install gazebo7
```
Once again check if the correct version was installed:
```
$ gazebo --version
```

diyjac, 2017-05-22:
It was an older 7.0 version in the VM.  I updated it to 7.7 following the README.md instructions and now am at 7.7.  That worked as can be seen here: https://files.slack.com/files-pri/T4ET9BQ3U-F5G7ZT2JZ/download/vm-upgrade-to-gazebo-7.7-fixed-camera.mp4

diyjac, 2017-05-22:
We may wish to have the students update their VM's gazebo before starting the arm_mover lesson.

hvpandya, 2017-05-22:
Glad it worked out!
Since we are planning to provide an updated VM to the students, I will go ahead and close this issue.

closed 2017-05-22 by hvpandya