udacity/robotics-beta-test-issues #365: P2 - Robotic arm - Pick & Place project - Missing instructions

diyjac, 2017-05-21:
Looking through the project repository at: https://github.com/udacity/test_repo_robond_robotic_arm_pick_and_place, I believe the following instructions maybe missing:

**NOTE: **: May want to move the README.md up to the top of the repository so it is easier to follow than down in `kinematics_project`.

1. How to clone the project and place it in its intended location:
```
$ mkdir ~/ros_ws
$ cd ~/ros_ws
$ git clone https://github.com/udacity/test_repo_robond_robotic_arm_pick_and_place.git src
$ catkin_make # some errors will be shown - this is to generate the devel directory.
```
2. Update the README.md instructions from:
```
$ cd ~ros_ws
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
$ cd ~ros_ws/src/kinematics_project/kuka_arm/scripts
$ sudo chmod +x target_spawn.py
$ sudo chmod +x IK_server.py
$ sudo chmod +x safe_spawner.sh
```
to
```
$ cd ~/ros_ws
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic -y
$ cd ~/ros_ws/src/kinematics_project/kuka_arm/scripts
$ sudo chmod +x target_spawn.py
$ sudo chmod +x IK_server.py
$ sudo chmod +x safe_spawner.sh
```
3. Make the nodes
```
$ cd ~/ros_ws
$ catkin_make
$ source devel/setup.bash
```
4. Launch Demo, instruction should be changed from:
```
$ cd ~ros_ws/src/kinematics_project/kuka_arm/scripts
$ ./safe_spawner
```
to
```
$ cd ~/ros_ws/src/kinematics_project/kuka_arm/scripts
$ ./safe_spawner.sh
```
**NOTE: This setup maybe too big for the VM in terms of screen size.**

[image]

hvpandya, 2017-05-21:
1. Fixed, thanks

2. Fixed, thanks

3. Fixed, thanks

4. Fixed, thanks

About your error, try:

```sh
$ locate Inertial.hh
```
and let me know what it says. Try to install the [ignition math package](http://ignitionrobotics.org/libraries/math), though you should already have it.

diyjac, 2017-05-21:
Hi @hvpandya!

Interesting, it appears that gazebo-7 has that include file in a different path:
```
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws/src/kinematics_project/kuka_arm/scripts$ locate Inertial.hh
/usr/include/gazebo-7/gazebo/physics/Inertial.hh
```
What would you like me to try?

hvpandya, 2017-05-22:
Interesting indeed, try to install ignition math and rebuild the project after that. 

This would have been easy to debug if **locate** command returned nothing .

Out of curiosity, did you update your gazebo-7 to 7.7 fro source?

diyjac, 2017-05-22:
No - I just follow the instructions from the README.md, so that update was from the apt-get repository.

diyjac, 2017-05-22:
Hmmm - it appears that I am unable to install `ignition math package`.  I use the link that you sent and this was the results:
```
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ wget -O /tmp/gazebo5_install.sh http://osrf-distributions.s3.amazonaws.com/gazebo/gazebo5_install.sh; sudo sh /tmp/gazebo5_install.sh
--2017-05-22 00:27:26--  http://osrf-distributions.s3.amazonaws.com/gazebo/gazebo5_install.sh
Resolving osrf-distributions.s3.amazonaws.com (osrf-distributions.s3.amazonaws.com)... 52.216.82.96
Connecting to osrf-distributions.s3.amazonaws.com (osrf-distributions.s3.amazonaws.com)|52.216.82.96|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1567 (1.5K) [application/x-sh]
Saving to: ‘/tmp/gazebo5_install.sh’

/tmp/gazebo5_install.sh  100%[==================================>]   1.53K  --.-KB/s    in 0s      

2017-05-22 00:27:27 (38.7 MB/s) - ‘/tmp/gazebo5_install.sh’ saved [1567/1567]

[sudo] password for jchen: 
This script will only work on Ubuntu trusty or utopic
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add - 
--2017-05-22 00:28:17--  http://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... 54.193.183.180
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|54.193.183.180|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1772 (1.7K) [application/pgp-keys]
Saving to: ‘STDOUT’

-                        100%[==================================>]   1.73K  --.-KB/s    in 0s      

2017-05-22 00:28:17 (52.9 MB/s) - written to stdout [1772/1772]

OK
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ sudo apt-get update; sudo apt-get install libignition-math-dev 
Hit:1 http://us.archive.ubuntu.com/ubuntu xenial InRelease
Hit:2 http://packages.ros.org/ros/ubuntu xenial InRelease                                          
Hit:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial InRelease
Hit:4 https://packagecloud.io/github/git-lfs/ubuntu xenial InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libignition-math-dev
```
Seems that the script was meant for Ubuntu `trusty` or `utopic`.

diyjac, 2017-05-22:
I have downloaded the source and am compiling it now.  Will know in a few minutes.

diyjac, 2017-05-22:
That did not work.  Installing the `ignite math package` manually installs it in `/usr/local/include/ignition/math/...` it seems.
```
[ 85%] Building CXX object src/CMakeFiles/UNIT_RotationSpline_TEST.dir/RotationSpline_TEST.cc.o
[ 87%] Linking CXX executable UNIT_RotationSpline_TEST
[ 87%] Built target UNIT_RotationSpline_TEST
Scanning dependencies of target UNIT_Pose_TEST
[ 89%] Building CXX object src/CMakeFiles/UNIT_Pose_TEST.dir/Pose_TEST.cc.o
[ 91%] Linking CXX executable UNIT_Pose_TEST
[ 91%] Built target UNIT_Pose_TEST
Scanning dependencies of target UNIT_Rand_TEST
[ 93%] Building CXX object src/CMakeFiles/UNIT_Rand_TEST.dir/Rand_TEST.cc.o
[ 95%] Linking CXX executable UNIT_Rand_TEST
[ 95%] Built target UNIT_Rand_TEST
Scanning dependencies of target UNIT_Triangle_TEST
[ 97%] Building CXX object src/CMakeFiles/UNIT_Triangle_TEST.dir/Triangle_TEST.cc.o
[100%] Linking CXX executable UNIT_Triangle_TEST
[100%] Built target UNIT_Triangle_TEST
Install the project...
-- Install configuration: "RelWithDebInfo"
Scanning dependencies of target man
Built target man
-- Installing: /usr/local/include/ignition/math/config.hh
-- Installing: /usr/local/lib/pkgconfig/ignition-math.pc
-- Installing: /usr/local/lib/cmake/ignition-math/ignition-math-config.cmake
-- Installing: /usr/local/lib/libignition-math.so.1.0.0
-- Installing: /usr/local/lib/libignition-math.so.1
-- Installing: /usr/local/lib/libignition-math.so
-- Installing: /usr/local/include/ignition/math.hh
-- Installing: /usr/local/include/ignition/math/AffineException.hh
-- Installing: /usr/local/include/ignition/math/Angle.hh
-- Installing: /usr/local/include/ignition/math/Box.hh
-- Installing: /usr/local/include/ignition/math/Filter.hh
-- Installing: /usr/local/include/ignition/math/Helpers.hh
-- Installing: /usr/local/include/ignition/math/IndexException.hh
-- Installing: /usr/local/include/ignition/math/Kmeans.hh
-- Installing: /usr/local/include/ignition/math/Line2.hh
-- Installing: /usr/local/include/ignition/math/Matrix3.hh
-- Installing: /usr/local/include/ignition/math/Matrix4.hh
-- Installing: /usr/local/include/ignition/math/Plane.hh
-- Installing: /usr/local/include/ignition/math/Pose3.hh
-- Installing: /usr/local/include/ignition/math/Quaternion.hh
-- Installing: /usr/local/include/ignition/math/Rand.hh
-- Installing: /usr/local/include/ignition/math/RotationSpline.hh
-- Installing: /usr/local/include/ignition/math/Spline.hh
-- Installing: /usr/local/include/ignition/math/Triangle.hh
-- Installing: /usr/local/include/ignition/math/Vector2.hh
-- Installing: /usr/local/include/ignition/math/Vector3.hh
-- Installing: /usr/local/include/ignition/math/Vector4.hh
```
Which produces this in the `catkin_make`:
```
. . .
[ 66%] Building CXX object kinematics_project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o
[ 75%] Building CXX object kinematics_project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o
[ 91%] Built target trajectory_sampler
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/jchen/RoboND/ros_ws/src/kinematics_project/gazebo_grasp_plugin/src/GazeboGraspFix.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/jchen/RoboND/ros_ws/src/kinematics_project/gazebo_grasp_plugin/src/GazeboGraspGripper.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
. . .
```

diyjac, 2017-05-22:
Wow - this is rather interesting!  It seems that the `ignition math package` no longer includes `ignition/math/Inertial.hh`:
```
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ ls /usr/local/include/ignition/math/
AffineException.hh  Filter.hh           Line2.hh            Pose3.hh            Spline.hh           Vector4.hh
Angle.hh            Helpers.hh          Matrix3.hh          Quaternion.hh       Triangle.hh         
Box.hh              IndexException.hh   Matrix4.hh          Rand.hh             Vector2.hh          
config.hh           Kmeans.hh           Plane.hh            RotationSpline.hh   Vector3.hh          
```

diyjac, 2017-05-22:
This must have been somehow done some time when gazebo was upgraded to 7.7 from 7.0 perhaps?

diyjac, 2017-05-22:
OK- have it working finally.  Had to clone it from bitbucket, and not use the archive.  Let me know if you see any issues.
```
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ rm -r build/ devel/
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ ls
activateROS.sh  src
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ catkin_make
Base path: /home/jchen/RoboND/ros_ws
Source space: /home/jchen/RoboND/ros_ws/src
Build space: /home/jchen/RoboND/ros_ws/build
Devel space: /home/jchen/RoboND/ros_ws/devel
Install space: /home/jchen/RoboND/ros_ws/install
####
#### Running command: "cmake /home/jchen/RoboND/ros_ws/src -DCATKIN_DEVEL_PREFIX=/home/jchen/RoboND/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jchen/RoboND/ros_ws/install -G Unix Makefiles" in "/home/jchen/RoboND/ros_ws/build"
####
-- The C compiler identification is GNU 5.3.1
-- The CXX compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/jchen/RoboND/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jchen/RoboND/ros_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/jchen/RoboND/ros_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.11") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jchen/RoboND/ros_ws/build/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - gazebo_grasp_plugin
-- ~~  - kuka_arm
-- ~~  - kr210_claw_moveit
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'gazebo_grasp_plugin'
-- ==> add_subdirectory(kinematics_project/gazebo_grasp_plugin)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
[... 255 lines cut ...]
Scanning dependencies of target kuka_arm_generate_messages_lisp
[ 25%] Generating Lisp code from kuka_arm/CalculateIK.srv
[ 25%] Built target kuka_arm_generate_messages_lisp
[ 25%] Built target kuka_arm_generate_messages_nodejs
Scanning dependencies of target kuka_arm_generate_messages_eus
Scanning dependencies of target kuka_arm_generate_messages_py
[ 41%] Generating Python code from SRV kuka_arm/CalculateIK
[ 41%] Generating EusLisp code from kuka_arm/CalculateIK.srv
[ 50%] Building CXX object kinematics_project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspFix.cpp.o
[ 58%] Generating EusLisp manifest code for kuka_arm
[ 66%] Generating Python srv __init__.py for kuka_arm
[ 66%] Built target kuka_arm_generate_messages_py
[ 75%] Building CXX object kinematics_project/gazebo_grasp_plugin/CMakeFiles/gazebo_grasp_fix.dir/src/GazeboGraspGripper.cpp.o
[ 75%] Built target kuka_arm_generate_messages_cpp
[ 75%] Built target kuka_arm_generate_messages_eus
Scanning dependencies of target kuka_arm_generate_messages
Scanning dependencies of target trajectory_sampler
[ 75%] Built target kuka_arm_generate_messages
[ 83%] Building CXX object kinematics_project/kuka_arm/CMakeFiles/trajectory_sampler.dir/src/trajectory_sampler.cpp.o
[ 91%] Linking CXX shared library /home/jchen/RoboND/ros_ws/devel/lib/libgazebo_grasp_fix.so
[ 91%] Built target gazebo_grasp_fix
[100%] Linking CXX executable /home/jchen/RoboND/ros_ws/devel/lib/kuka_arm/trajectory_sampler
[100%] Built target trajectory_sampler
(ros_ws) jchen@jchen-Inspiron-15-7568:~/RoboND/ros_ws$ 
```

diyjac, 2017-05-22:
This is also an issue in the VM.
```
. . .
[ 75%] Building CXX object kinematics_project/kuka_arm/CMakeFiles/trajectory_sampler.dir/src/trajectory_sampler.cpp.o
In file included from /usr/include/gazebo-7/gazebo/gazebo_core.hh:21:0,
                 from /usr/include/gazebo-7/gazebo/gazebo.hh:20,
                 from /home/robo/ros_ws/src/kinematics_project/gazebo_grasp_plugin/src/GazeboGraspFix.cpp:2:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
. . .
```
[image]

diyjac, 2017-05-22:
Ok - got it to compile by cloning the bitbucket repository:
```
$ hg clone https://bitbucket.org/ignitionrobotics/ign-math
$ cd ign-math
$ mkdir build
$ cd build
$ cmake ..
$ sudo make install
```
But also have to create a symbolic link to the math directory because it is now two levels deeper:
```
$ cd /usr/local/include/ignition
$ ls
math4
$ sudo ln -s math4/ignition/math .
$ ls -l
total 4
lrwxrwxrwx 1 root root   19 May 22 11:39 math -> math4/ignition/math
drwxr-xr-x 3 root root 4096 May 22 11:38 math4
```

[image]

P2 running in VM:
[image]

hvpandya, 2017-05-22:
I just created a fresh VM, and reproduced this error. I was able to resolve it by simply:
```sh
$ sudo apt-get install libignition-math2
```

diyjac, 2017-05-22:
Cool!  That's certainly a lot simpler!

closed 2017-05-23 by hvpandya