udacity/robotics-beta-test-issues #366: qiang ROS Essentials L02E03

luqiang21, 2017-05-21:
I can't paste the git clone command in the robo-nd virtual machine. I am using Mac.

bkinman, 2017-05-22:
Are you running VMWare under OSX?

luqiang21, 2017-05-22:
@bkinman Yes. I found another way, just opened udacity webpage inside the VMware.

bkinman, 2017-05-22:
This is an issue that I've run into multiple times over the years when using VMs. It usually has something to do with the drivers you've installed in the host machine, or the lack of the guest drivers in the virtual machine.

I'm going to go ahead and leave this open for now; It's an issue that should be mentioned in our VM faq document, which has not yet been created.

luqiang21, 2017-05-24:
Update: I used the latest vmdk, and was able to copy and paste between Mac and the robo-nd VM.

bkinman, 2017-05-24:
Excellent! Good to hear! still going to leave this open to we know to make a FAQ entry!

kylesf, 2017-06-10:
@bkinman What are your thoughts for this FAQ?

sahiljuneja, 2017-06-10:
@kylesf I'm working on one currently. Will be sharing it with beta testers to add more to it soon.

kylesf, 2017-06-10:
Awesome, I will assign this to you then.

closed 2017-11-21 by kylesf