udacity/robotics-beta-test-issues #367: P2 - Robotic arm - Pick & Place project - Missing shelf and dropbox in Gazebo

diyjac, 2017-05-21:
Once the Gazebo and rviz are up and running in demo, I do not see the shelf in front of the robot or dropbox to its right, so I am reporting it as requested in the README.md project file.  There is a cube in front of the robot instead with the blue cylindrical target on the floor instead of the missing shelf or the cube.

[image]

hvpandya, 2017-05-21:
Your env variable pertaining to gazebo is not set properly. 

You need to modify following line from README as it applies to your directory structure:
```
export GAZEBO_MODEL_PATH=~/ros_ws/src/kinematics_project/kuka_arm/models
```

diyjac, 2017-05-21:
Hi @hvpandya!

It appears that was the problem.  I am able to see the self and the dropbox now.

Cheers!

[image]

closed 2017-05-21 by diyjac