udacity/robotics-beta-test-issues #376: crl L03E03  Add a package

zenetio, 2017-05-22:
I did not got the controller-manager error mentioned in the lesson. Instead, I got a warning message regarding to joint_1 and joint_2 that I had to update in simple_arm.urdf.xacro

bkinman, 2017-05-22:
What warning message did you get, and what did you update in simple_arm.urdf.xacro?

Does your system have `ros-kinetic-controller-manager` on it (`dpkg --list | grep controller`)?
Are you installed natively or using the virtual machine?

zenetio, 2017-05-22:
Yes, I have ros-kinetic-controller-manager installed. 
Below is the message I got. Then to fix it I just prepended the string hardware_interface before EffortJointInterface, I mean, for joint_1 and joint_2 names, I edited the simple_arm/urdf/simple_arm.urdf.xacro file as follow:

changed from 
<joint name='joint_1">
<hardwareInterface>EffortJointInterface</hardwareInterface>
</loint>
to
<joint name='joint_1">
<hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
</loint>

[WARNING] [1495495164.509320009, 261.179000000]: No matching hardware interface found for '/EffortJointInterface' while loading interfaces for joint_1
[WARNING] [1495495164.509548097, 261.179000000]: Could not initialize robot simulation interface

jchernus, 2017-05-23:
Can confirm, ros-kinetic-controller-manager is already installed. No error message present when running catkin_make in the VM.

zenetio, 2017-05-23:
Below is the correct warning message I got and after follow the instructions in the message, the warning message was fixed. But I am still getting a weird behavior: if I edit the .xacro file as suggested by the warning message, I don't get any warning or error message, but the arm does not move. By the other side, if I force an error during edition writing only a slash '/' instead of hardware_interface/, I get a FATAL ERROR saying that "Could not initialize robot simulation interface", when I run roslaunch but the arm starts moving as showed in the video in the lesson.

[ WARN] [1495514164.096750213, 261.179000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'joint_1'.

bkinman, 2017-05-23:
ros-kinetic-controller-manager should not be installed in the VM. If you've run any of the arm code from the Kinematics lesson, then it would have been necessary to install ros-kinetic-controller-manager. Is it possible that this is how you guys ended up with this package installed?

bkinman, 2017-05-23:
In terms of the warning, I'm going to go ahead and leave that alone for now. The warning merely indicates that something is deprecated... We have some time to figure out a good fix for it.

closed 2017-05-23 by bkinman