udacity/robotics-beta-test-issues #383: ROS Essentials - L02E05 Rosdep

jchernus, 2017-05-23:
Using: VM

`rosdep check simple_arm` comes up with 3 missing dependencies. Video only shows 1. 

Not a big problem, since `rosdep install -i simple_arm` installs all 3 missing packages, but students might be confused/worried.

apt	ros-kinetic-gazebo-ros-control
apt	ros-kinetic-joint-state-controller
apt	ros-kinetic-effort-controllers

bkinman, 2017-05-23:
Hrmm. This is a good point. I'll create an issue to update this in the VM.

bkinman, 2017-05-23:
Actually, I'll just reopen this one, and give this the VM label.

jchernus, 2017-05-23:
Warning: I wasn't using today's VM. This might have been fixed in the latest one.

bkinman, 2017-05-24:
Note: to make the latest VM  (5/22) agree with the lecture, install `ros-kinetic-effort-controllers`

kylesf, 2017-06-10:
Okay, checked and fixed. Will be in the new VM. Will put in progress until a new successfully VM is released.

kylesf, 2017-06-12:
This has been taken care of and you can find the updated VM in the classroom! Thanks for the help!

closed 2017-06-12 by kylesf