udacity/robotics-beta-test-issues #384: Vivek 1108 DH Parameters.

vxy10, 2017-05-23:
Why DH parameters? Can we avoid them? DH parameters are prone to error and unnecessarily complex. Exponential matrices are better. Although, to appreciate exponential matrices, one needs to suffer through DH parameters. So its ok to have that too. But recommend introducing both.

thomas-at-em, 2017-05-23:
I thought about that and ultimately decided not to. 2 weeks are devoted to kinematics with the expectation that students would spend ~15 hours/week studying. Ultimately that constrains the amount of material that you can present. I would agree with you that in many respects the product of exponentials formulation is better than DH parameters, but DH parameters are far more common. Manufacturer's of serial manipulators typically publish DH parameters in their product brochures or user manuals.

closed 2017-05-23 by thomas-at-em

vxy10, 2017-05-23:
It will be nice to at least point to materials and mention other methods.