udacity/robotics-beta-test-issues #393: P2 - Pick and Place [README]

jchernus, 2017-05-23:
git repo README.md:

1) Reads,

"Now that you have a workspace, clone or download this repo into the src directory of your workspace and build it:

```
$ cd ~/ros_ws/src
$ git clone https://github.com/udacity/test_repo_robond_robotic_arm_pick_and_place.git
```"

This will produce an error. Do students need to build it right away, can they wait until they complete the next step?

2) Now, in the next step,

```
$ cd ~/ros_ws/src/kinematics_project/kuka_arm/scripts
```

won't work because the directory structure after the git clone command is 

```
~/ros_ws/src/test_repo_robond_robotic_arm_pick_and_place/kinematics_project/kuka_arm/scripts
```

Maybe add a mv command in there to move kinematics_project to src directly?

3) Also, I had to `sudo apt-get install libignition-math2` for the build to work, but I'm sure someone has mentioned this already.

hvpandya, 2017-05-23:
Fixed the directory issue.

The new VM already contains ignition-math2
@jchernus please use the latest VM image for testing.

closed 2017-05-23 by hvpandya

jchernus, 2017-05-23:
Switching to newest VM. Thanks.