udacity/robotics-beta-test-issues #394: Current (5/23/17) VM Issues

kylesf, 2017-05-23:
- Issue #1

`roslaunch simple_arm robot_spawn.launch` runs fine before installing missing dependencies but after installing missing dependencies via `rosdep install -i simple_arm` and running `roslaunch simple_arm robot_spawn.launch` its crashes with segmentation fault

- Issue #2

Can we use `sudo apt autoremove` to remove all of the unneeded packages? Or can it be done prior to shipping VM. It is very annoying when trying to read any runtime actions of `sudo apt-get`

kylesf, 2017-05-23:
Seems I have issue #1 working. Not sure if continuing actions in the layout of the course resolved it or if there was a mistake on my end. Waiting on other students to share their experiences with new VM, instead of me starting from scratch again to check .

alvarovillena, 2017-05-24:
I can confirm that everything is working fine on my VM, so issue #1 is actually not an issue :)

However,  [this lesson](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/ae288087-b8ef-4e2d-97ff-a85993300ac7/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/658c94f5-f806-4273-9001-9e2838e56856/concepts/ad9785ee-a561-4a98-bf11-a8ef32d6fc75) says that before installing dependencies the following warning message should appear when running `roslaunch simple_arm robot_spawn.launch`:
```The controller spawner couldn’t find the expected controller_manager ROS interface.```
This message appeared when using the old VM, but now I'm getting error messages instead:
```
[ERROR] [1495616010.582090901, 261.712000000]: Could not load controller 'joint_1_position_controller' because controller type 'effort_controllers/JointPositionController' does not exist.
[ERROR] [1495616010.583152246, 261.712000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1495616011.587968, 262.693000]: Failed to load joint_1_position_controller
[ERROR] [1495616011.593300466, 262.699000000]: Could not load controller 'joint_2_position_controller' because controller type 'effort_controllers/JointPositionController' does not exist.
[ERROR] [1495616011.593389505, 262.699000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1495616012.595061, 263.670000]: Failed to load joint_2_position_controller
```

Even with this messages, the program is working as expected. Probably I did something wrong, it would be nice if someone else can check this.

hvpandya, 2017-05-24:
Issue #1: Seems resolved

Issue #2: How annoying would this be to an average student in your opinion? 
The reason we have installed the full desktop version of ROS in VM, is to allow students to play around on their own in the full fledged ROS environment outside of this program. For example if a student is done with ROS Essentials and wants to play with various sensors on the simple_arm outside of this program, they should have most tools readily available to do so.

Issue #3: +1
Since we use nano in some of the screencasts, it makes more sense to do this for nano instead.

Issue #4: +1 
@bkinman can we add this extraction command in the VM concept?

Suggestion #1: I feel this one is very subjective and based on ones personal preferences. We will leave this for students to decide.

Suggestion #2: I do not see how a single `apt-get install git` command will add to student learning. We already have several `apt-get install` commands throughout this program for students to get habituated with.

Suggestion #3: I like this suggestion, @ryan-keenan is this something we can add to the **every 3rd week** content?

kylesf, 2017-05-24:
1 - Done
2 - Sounds good, I was under the impression that they were not used at all.
3 - Only reason I said vim is because when you run the classroom command you get an error that vim is not installed. Perhaps there is another way to open them with nano
4 - Good

1- Sounds good and no worries just sharing (I am okay with being wrong :) )
2 - Sounds good 
3 - Cool

@hvpandya Thanks for your time. Whenever the team is done with this you guys can close it. I will not be adding anything more to this one

bkinman, 2017-05-24:
In the future, can we file these things as individual issues?

kylesf, 2017-05-24:
@bkinman We were asked not too by @hvpandya 

```
A quick request to anyone testing the Pick and Place project, if you find typos or other problems in the README file, file a single ticket with all your findings. You may even reopen a closed typo ticket and add new findings. 
This will help us reduce the clutter on waffle board :slightly_smiling_face:

P.S. if you reopen a ticket, tag the assignee in comments.

```

bkinman, 2017-05-24:
Issue 1: non-issue
Issue 2: wont' implement change
Issue 3: Creating a new issue for this, electing to set default editor to nano, script updated to reflect that we could re-record at some point.
Issue 4: instructions added.
Suggestion 1: Ticket created
Suggestion 2: Now that git is included in the VM, i'll just remove the instruction telling students to install it. I don't think installing git helps with student learning.
Suggestion 3: With Unix being a pre-requisite for the course I'm expecting students will understand how to edit files from the command-linux

bkinman, 2017-05-24:
@kylesf Harsh was speaking with respect to README problems in the Pick and Place project.

closed 2017-05-24 by bkinman

kylesf, 2017-05-24:
Got it thanks.