udacity/robotics-beta-test-issues #398: Kinematics L01E03

jchernus, 2017-05-23:
**Nitpicks:**

1) "In the previous example, we saw that two particles connected by an inextensible rod and confined to a planar surface did not have 4 degrees of freedom, only three."

change '4' to 'four'

2) You use _'DoF'_ after the image. Typically, you should say _"degrees of freedom (DoF)"_ so that it is super clear.

3) Can you expand this,

"DoF = 6(number of moveable links) - 5(number of 1-DoF joints) DoF = 6(3) - 5(3) = (6 - 5) * 3 = 3"

to read as such,

Total DoF = 6(number of moveable links) - 5(number of 1-DoF joints) 
= 6(3) - 5(3) 
= 18 - 15
= 3

for better readability.

thomas-at-em, 2017-05-23:
Fixed

closed 2017-05-23 by thomas-at-em

hvpandya, 2017-05-24:
@jchernus @thomas-at-em I would however like to point that DOF is the unofficially official abbreviation used in ROS

thomas-at-em, 2017-05-24:
I can live with that discrepancy.