udacity/robotics-beta-test-issues #400: Generalized coordinates

vxy10, 2017-05-24:
This statement is not complete. "Calling the variable a generalized coordinate helps illustrate the fundamental point that the variable could be an angle, or distance, or even something else of devoid of geometric significant" 

Generalized coordinates have one important property, "They are minimal set of joint angles needed to uniquely define the robotic system" Also, choice of generalized coordinates are not unique, but choosing them wisely can greatly simplify the math. The second part must be included in lecture

thomas-at-em, 2017-05-24:
I don't understand why you say the quoted statement is "not complete". Please explain.

Generalized coordinates are not necessarily a minimal set; however, for a holonomic system (not discussed) you can always find a set of generalized coordinates that is a minimal set, i.e., requires no constraint equations.

vxy10, 2017-05-24:
my bad, incomplete may be wrong wording. i meant to say that additional information can be incorporated too. Thanks.

closed 2017-05-29 by thomas-at-em