udacity/robotics-beta-test-issues #403: Write ROS Nodes - Command to set permissions changes

alvarovillena, 2017-05-24:
Along this lesson, the following command is used to set execution permissions on a file: `chmod u+x filename`. The command used [here](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/ae288087-b8ef-4e2d-97ff-a85993300ac7/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/ade14c05-008c-4802-bfd8-f0271db9facc/concepts/faa4d5c5-54e9-4e35-9fb1-08e17b2bd97e) to change permissions for `hello`script is slightly different: `chmod +x hello`

Both commands are very similar, but the second one set execution permissions for everyone, while the first one just give this permission to the user. I think that the same command should be used everywhere to avoid confusion.

hvpandya, 2017-05-24:
+1 lets just use chmod +x filename

bkinman, 2017-05-24:
Harsh and I decided upon u+x, closing issue.

hvpandya, 2017-05-26:
Made the changes in my scripts. @bkinman you still need to make one change suggested in the OP.

bkinman, 2017-06-02:
Fixed!

closed 2017-06-02 by bkinman