udacity/robotics-beta-test-issues #404: Kinematics - L01E11 [Typo]

jchernus, 2017-05-24:
You have seen how to construct rotation matrices by hand, I will now show you how to do this in code with a Python package called SymPy. SymPy is a full-featured computer algebra system (CAS) that will enable you to construct and **manipulator** matrices symbolically and then numerically evaluate them when needed.

manipulator -> manipulate?

jchernus, 2017-05-24:
Another: 

With rotations most **function** expect angles to be input as radians, but for most people units of degrees **is** more intuitive. Defining reusable conversion factors is generally a good idea.

function -> functions
is -> are?

Consider: 

For rotations, most functions will expect angles to be input as radians. However, most people find units of degrees to be more intuitive. Defining reusable conversion factors is often a good idea.

thomas-at-em, 2017-05-24:
Good catch, thanks. If possible, please include the lesson name in your feedback because it makes it easier for me to find the problem. Keep up the good work!

closed 2017-05-24 by thomas-at-em

jchernus, 2017-05-24:
Sorry, will do!