udacity/robotics-beta-test-issues #405: crl L03E05 Arm Mover: The code

zenetio, 2017-05-24:
This is a list of issues I've found in lesson 03 session 05
1. There are some blank lines separing some statements which make code visialization a bit confuse
2. In mover_service(), remove 2 blank lines and a colon is missing before Float64 for both j1_publisher and j2_publisher
3. In The code: explained, I see (e.g. ~min_joint_1_angle resolves to /simple_arm/max_joint_1_angle) when it should before
(e.g. ~min_joint_1_angle resolves to /simple_arm/min_joint_1_angle)
4. There is an additional the in Warning messages are logged if the the requested
5. In the comment above handle_safe_move_request I read handle_service_request when it should be handle_safe_move_request
6. At the end of the page where I can see the if __name__ , the j1_publisher and j2_publisher are incomplete

bkinman, 2017-05-25:
Are you speaking out the code in the classroom, or the code in the github repo? I'm assuming that you mean the code in the classroom.

1. I removed a single blank line that was extra.
2. in mover_service(), the publisher lines are wrapping funny due to the way the markdown text is being formatted. I did a modification that should make the line wrapping behave better, but unfortunately I don't think there's an easy fix for this. **Also, I'm not sure what you mean by colon? Are you speaking with respect to a comma?**
3. fixed!
4. **If the request what?**
5. Awesome catch, fixed!
6. **Can I get a screenshot? It seems to display properly on my system**

zenetio, 2017-05-26:
4. There are 2 "the" words in the sentence, "the the requested".
6. Its ok now. Fixed!

closed 2017-06-02 by bkinman