udacity/robotics-beta-test-issues #418: DH parameters example (Vivek1108)

vxy10, 2017-05-25:
Hi, will be nice to include example of DH conventions. You can give link to section 3.2 here. Section 3.3 has many examples too. It explains really well, http://smpp.northwestern.edu/savedLiterature/Spong_Textbook.pdf

thomas-at-em, 2017-05-25:
Yes, I have decided to add a screencast of a DH parameter example. The Spong book you reference uses a different convention than the one I teach.

closed 2017-05-25 by thomas-at-em

vxy10, 2017-05-25:
Thanks, i was going to add that ticket too. Any reason for using the convention you did? Is it more common among people using ROS framework?

thomas-at-em, 2017-05-25:
Sort of. I personally find it to be more intuitive than the original convention used by Spong. It does have the advantage that, if you follow the rules, ALL the constant terms in the first and last rows of the DH table are zero. So it does result in (modestly) simpler equations.

vxy10, 2017-05-26:
Thanks, i didnt know of this. I did robotics courses 10 years ago. Btw, will be nice to include in lecture that these conventions are different. As others may get confused too.