udacity/robotics-beta-test-issues #419: Didactic suggestion: Joint Types & Degrees of Freedom calculation too soon

letyrodridc, 2017-05-25:
I was following the explanation about calculating the DoF quantity in each joint, suddenly I've read this text:
"Thus, two rigid bodies in free space connected by a one DoF revolute joint would only have a total of 2*6 - 5 = 7 DoF. Note that for modeling purposes, higher order pairs can always be replaced by a collection of lower order pairs so without loss of generality, we can restrict our attention to only the revolute and prismatic types."

The calculation comes ahead from the formula. I couldn't see where that calculation comes from. Maybe, some additional explanation could be added before or that fragment could be moved forward.

thomas-at-em, 2017-05-25:
OK, I will try to make this point more clear.

kylesf, 2017-06-13:
Updated!

closed 2017-06-13 by kylesf

letyrodridc, 2017-07-29:
Thanks Kyle! That lessons changed a lot. Everything more clear.