udacity/robotics-beta-test-issues #422: Kinematics: 6. Coordinate Frames and Vectors Coordinate Frames and Vectors ||a||=1(?) missing (letyrodri)

letyrodridc, 2017-05-25:
Hi, following the lesson, I couldn't find a place where said that ||a|| is 1, because it's an unitary vector.

I think that it's an important data to have, and to understand later this formula 
v.a​​ =∥v∥∥1∥cos(θ)

I don't understand notation. The hat ​^ is uses to indicate vector. But v has no hat, and a has. 

Maybe, you want to indicate a with hat as an unitary vector, but I'm not sure that's everywhere standard.

I hope my comments help.

letyrodridc, 2017-05-25:
I found that b has the hat and modulus greater than 1 .... mm I sound like an obsessive hat person :D sorry.

letyrodridc, 2017-05-25:
The hat is minor, the ||a|| = 1 seems more important and conceptual.

thomas-at-em, 2017-05-25:
I can see how that is confusing. The "hat" is customarily used to denote unit vectors. Bold is for non-unit vectors.

letyrodridc, 2017-05-25:
Is *b* vector also unit vector? It has hat.

thomas-at-em, 2017-05-25:
Any quantity with a "hat" is a unit vector.

closed 2017-05-25 by thomas-at-em