udacity/robotics-beta-test-issues #424: Vivek 1108 composition of rotation matrices error.

vxy10, 2017-05-25:
There seems to be error in part where rotation composition is doen for R_ZYX in fixed frame. The final rotation matix must be,

```

    Matrix([
    [                    0,                     0, 1.0],
    [1.0*sin(alp)*cos(gam), 1.0*cos(alp)*cos(gam),   0],
    [1.0*sin(alp)*sin(gam), 1.0*sin(gam)*cos(alp),   0]])
```

Also, something similar to gimbal lock with z and x axis can be obtained using R_zyx body centered. I used, Rot_X(gam), Rot_Y(pi/2) and Rot_Z(alp) to get, 

R_zyx = Rot_alp*Rot_pi2*Rot_gam # POST MULTIPLY, NOT PRE

```
Matrix([
[0, -1.0*sin(alp - gam), 1.0*cos(alp - gam)],
[0,  1.0*cos(alp - gam), 1.0*sin(alp - gam)],
[0,                   0,                  0]])
```

vxy10, 2017-05-25:
the first matrix must be,
```
Matrix([
[                  0,                  0, 1.0],
[ 1.0*sin(alp + gam), 1.0*cos(alp + gam),   0],
[-1.0*cos(alp + gam), 1.0*sin(alp + gam),   0]])
```

Copied wrong one there, please ignote the first one

thomas-at-em, 2017-05-26:
Yes, this is correct. Thanks very much for catching it! It should be fixed now.

closed 2017-05-26 by thomas-at-em