udacity/robotics-beta-test-issues #447: arek - Project: Robotic Arm: Pick & Place - 8. Pick and Place Demo Walkthrough

867sec, 2017-05-28:
arek - Project: Robotic Arm: Pick & Place - 8. Pick and Place Demo Walkthrough
On the page the instructions say to change demo flag to true and they say: Once Gazebo and rviz are up and running, make sure you see the following objects in the gazebo world and the rviz window:

Robot
Shelf
Blue cylindrical target in one of the shelves (not present in rviz)
Dropbox right next to the robot

but the shelf and bucket only appears when the demo flag is set to false. Please see the screenshot attached when the demo flag is set to true below:

hvpandya, 2017-05-29:
This is the step that you need to perform properly for objects to spawn correctly:

Add following to your .bashrc file

export GAZEBO_MODEL_PATH=~/ros_ws/src/kinematics_project/kuka_arm/models

Make sure your directory structure resembles that of the instructions. Change the name of the workspace and the path to match that of yours.

867sec, 2017-05-29:
If this is the case that I have to add the path equals strating with
"~/ros_ws" so why in th Project:Robotic Arm:Pick and Place - 2: Setting up
your environment it says the following:

"For the rest of this setup, catkin_ws is the name of active ROS Workspace,
if your workspace name is different, change the commands accordingly."

Please see screenshot attached. On top you can see the text I mentioned and
at the bottom you can see GAZEBO_MODEL_PATH instructions with catkin.

My workspace is catkin_ws and I have in my bashrc "export
GAZEBO_MODEL_PATH=~/catkin_ws/src/kinematics_project/kuka_arm/models" so
why are you saying that I should have "export
GAZEBO_MODEL_PATH=~/ros_ws/src/kinematics_project/kuka_arm/models" in this
ticket since it's never mentioned in the course materials?

Please advise.

Best regards,
Arek Kaz

>
> This is the step that you need to perform properly for objects to spawn
correctly:
>
> Add following to your .bashrc file
>
> export GAZEBO_MODEL_PATH=~/ros_ws/src/kinematics_project/kuka_arm/models
>
> Make sure your directory structure resembles that of the instructions.
Change the name of the workspace and the path to match that of yours.

hvpandya, 2017-05-29:
@867sec What part of this 

> Make sure your directory structure resembles that of the instructions. Change the name of the workspace and the path to match that of yours.

is unclear to you? 

Whether your workspace name is catkin_ws or ros_ws, my instructions clearly state that you should change your **workspace** and **path** according to your setup. 

> why are you saying that I should have "export
GAZEBO_MODEL_PATH=~/ros_ws/src/kinematics_project/kuka_arm/models" in this
ticket since it's never mentioned in the course materials?

I did not, I just showed you which step is causing the problem. Again, I clearly stated in my answer that you should check your **ws** & **path** and make changes to that step accordingly.

From within your **src** directory, can you paste the results of `$ pwd` ?

hvpandya, 2017-05-29:
This is the same problem as #367

867sec, 2017-05-29:
/home/robond/catkin_ws/src

867sec, 2017-05-29:
This is no longer an issue per Harsh explanation.

closed 2017-05-29 by 867sec