udacity/robotics-beta-test-issues #464: sjuneja L1.3 Joint Types and DoF - There's no explanation on what DoF really is

sahiljuneja, 2017-05-30:
This was quite surprising.

The text just starts with talking about DoF as if the student knows what they are. But that wasn't covered in any lesson before, was it? 

---

```
As we saw in the previous lesson, joints constrain the motion of bodies and reduce the number of degrees of freedom.
```

I am assuming there's a missing lecture you guys will be adding which would cover this in detail? 

---

```
As shown in the above figure, joints can be grouped into lower order pairs having one DoF and higher order pairs offering two or three DoF. For serial manipulators (a.k.a., robot arm or kinematic chain), the most common joint types are revolute and prismatic allowing one rotational and one translational DoF, respectively. In other words, a one DoF joint destroys five DoF. Thus, two rigid bodies in free space connected by a one DoF revolute joint would only have a total of 2*6 - 5 = 7 DoF. Note that for modeling purposes, higher order pairs can always be replaced by a collection of lower order pairs so without loss of generality, we can restrict our attention to only the revolute and prismatic types.
```

This needs more explanation. 

- There's no explanation of what lower and higher order pairs are here.
- Explanation/visual for serial manipulators
- From the image they can probably figure out what revolute and prismatic are, but it requires some input for the students.
- The `joint destroys 5 DoF` is not intuitive.
- Explanation of what `generality` refers to here. 

Also, is the above required, honestly? I feel this could be diluted in terms of what you are trying to convey and the rest can be left out. Not sure, what do you think?

---

An explanation on a breakdown of the equation to calculate the DoF would be helpful.

---

Do you think the footnote is required here? I feel it's perhaps a bit much given we aren't going into more detail. Although, it's good as an added resource (as long as students can make the jump from the lesson to the resource without too much difficulty).

kylesf, 2017-07-28:
This still a floating issue?

sahiljuneja, 2017-07-28:
I think it's covered for now. Closing it.

closed 2017-07-28 by sahiljuneja