udacity/robotics-beta-test-issues #468: sjuneja L.17 D-H Parameters - Visuals and terminology

sahiljuneja, 2017-05-30:
I, honestly, don't recommend the current visual for explaining the parameters for the students. It's not intuitive and will confuse them.

---

Clarifying `right-hand sense` might be good.

---

```
Notice, as in the figure above, there is no requirement for the origin of the frame to be physically on the link, only that it move rigidly with the link. 
```

Don't you think this will confuse some students on how `d_i` is calculated? 

---

Specific examples of where the 4 cases of orientation of  `Z_i` w.r.t `Z_(i-1)` might help students visualize the problem better. Like for which robot perhaps? It would help even more if they could go step-by-step through these (or couple of these) cases.

jchernus, 2017-06-02:
+1. 

I saw a note somewhere that this graphic might have been sent to animators to be re-created. I hope that's the case. This one is NOT intuitive.

vxy10, 2017-06-03:
Adding a video explaining the process will help

Shreeyak, 2017-06-06:
The `right-hand sense` was mentioned a couple of times and I did get a little confused, until I went back and reviewed my materials. I think there should be a section explaining this rule in more detail. It is critical in the DH parameters when deciding if its pi/2 or -pi/2 and I'm sure will raise a *lot* of doubts if left unexplained.

jchernus, 2017-06-13:
"For an n-degree of freedom manipulator, there will be n-joints {1, 2, …, n} and, because every joint connects two links, there are n+1 links {0, 1, …, n}. " 

The image only shows 2 links (for 2 joints). I think it would be helpful to modify the image to include Link 0.