udacity/robotics-beta-test-issues #470: sjuneja P02.6 - Understanding URDF - Needs more explanation

sahiljuneja, 2017-05-30:
https://classroom.udacity.com/nanodegrees/nd209-beta/parts/72800973-80cd-4828-81fd-b3390486892b/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/91d017b1-4493-4522-ad52-04a74a01094c/concepts/304b7bc0-6fe8-4614-bd09-4a545665adad

If the students are expected to extract DH parameters from the URDF then they need to be able to figure out how to convert details from a URDF to the appropriate frame of references for the links/joints to calculate the DH parameters.

So I think this section should be expanded such that we take a smaller robot's URDF as an example, and step-by-step indicate how they can create reference frames based on the URDF. 

Otherwise, this lecture currently isn't particularly helpful from a student's perspective.

hvpandya, 2017-05-30:
I totally agree with you on this, but the idea was to use URDF as a reference and use the `forward_kinematics.launch` demo for actually getting the parameters.

I believe I can reword certain places to suggest that the urdf is meant only for reference but the main source of param extraction should be the FW demo.

Does that resolve this ticket?

sahiljuneja, 2017-05-30:
Yeah, rewording to indicate so would be good. But do you think being able to derive them using just the URDF is feasible and something they should be able to do from a practical and learning experience? You have more experience with this so will leave to your judgment!

The lesson was introduced today and I think that should be helpful. I am going through that now and going to work on the project. I'll create a separate issue if I find something.

Thanks!

Shreeyak, 2017-05-31:
@hvpandya The lesson specifically states that all we need is the URDF.  I had no idea that I needed to used the FK demo to extract params. 

Even so, this section needs more explanation. there's a description of the URDF given, but no figure of the robot to correlate to. Best would be to take the robot, and show how the urdf is built step-by-step. The URDF is a very important concept for any kind of simulation and I think this concept is worth elaborating. Students should know what an URDF is and how one is built using an example.

hvpandya, 2017-05-31:
@Shreeyak I am quite sure the use of FK demo has been stressed a lot in different places. 

The idea here is to introduce only the basics of URDF and how simple kinematics fits into URDF. This is by no means an attempt to "teach" URDF, this is only meant to help people parse through the file for project purpose. Remember that this specific lesson is only meant for the project and not a learning lesson.

Extensive robot modelling will most likely be covered in term 2.

Shreeyak, 2017-05-31:
Oh, OK that's great to hear. if this point was stressed in the lesson at
that point it would be nice. I myself was misled into thinking I was to use
the urdf, because of the statement I mentioned earlier.

Shreeyak, 2017-06-06:
@hvpandya Are the students expected to get link lengths also from the  forward_kinematics.launch demo ??

kylesf, 2017-06-13:
Do we agree that the content in the classroom now allows for students to successfully extract values from urdf?

sahiljuneja, 2017-06-14:
Upcoming content will cover how URDF can be used for obtaining values for FK.  I think we can close this for now.

closed 2017-06-14 by sahiljuneja