udacity/robotics-beta-test-issues #471: Kinematics - L01C12 Composition of Rotations

jchernus, 2017-05-30:
The two programming quizzes at the bottom - 

Quiz A) Asks us to intrinsically rotate about Y & then Z.

**YZ_intrinsic_sym = R_y * R_z**

Quiz B) Asks us to extrinsically rotate about Z then Y. 

Shouldn't **ZY_extrinsic_sym = R_y * R_z**?

I also think you should be asking students to rotate about the same axes both times - not changing them up. This just adds unnecessary confusion.

Shreeyak, 2017-05-31:
I skipped this part of the lecture. While reading this, it completely skipped past me that in B) the axis has changed. It would be better to keep the same axis in both as a manner of seeing the difference in the result.

thomas-at-em, 2017-05-31:
Quiz 3 attempts to explain why I choose quiz 1 and quiz 2 the way I did.

closed 2017-05-31 by thomas-at-em