udacity/robotics-beta-test-issues #472: Kinematics - L01C13 Euler Angles from Rotation Matrix

jchernus, 2017-05-30:
"In many cases, particularly involving inverse kinematics, we are given a composite rotation matrix and it is necessary to find a set of Euler Angles that would produce this rotation." 

Define such a case. Where do you get the composite rotation matrix from if you didn't start with the angles? Provide an example please.

thomas-at-em, 2017-05-31:
You encounter this problem anytime you are interested in solving inverse kinematics (IK). In the IK problem you are assuming you know the pose (i.e., position and orientation) of the end effector and you want to find the joint angles to produce this pose. Specifically, students will encounter this in the pick and place project. When they issue the command rosrun tf tf_echo \base_link \gripper, ROS will return position and orientation of the end effector relative to the base link. As mentioned in "Composition of Rotations" lesson, Euler's Theorem states that any orientation of a rigid body can be composed of three elementary rotations (i.e., Euler Angles).

kylesf, 2017-07-28:
@jchernus Did you want to add this to the classroom or close the issue?