udacity/robotics-beta-test-issues #473: qiang Introduction to Kinematics-Composition of Rotations

luqiang21, 2017-05-31:
For the two quizzes, I don't know what is wrong when I run the solution code, following error:

raceback (most recent call last):
  File "vm_main3.py", line 47, in <module>
    import main
  File "/tmp/vmuser_umxydwhwen/main.py", line 2, in <module>
    import studentMain
  File "/tmp/vmuser_umxydwhwen/studentMain.py", line 5, in <module>
    print('Composite YZ rotation is: ',rotation_template.YZ_intrinsic_num)
AttributeError: 'module' object has no attribute 'YZ_intrinsic_num'

thomas-at-em, 2017-05-31:
I am unable to reproduce your error. If I simply copy/paste lines 14 to 22 of the solution.py into rotation_template.py (replacing lines 14 to 17) and hit "Test Run", I get numerical output I would expect. If I hit "Submit" I get "Nice Work. Looks Good!" This is true for both quizzes.

luqiang21, 2017-05-31:
@thomas-at-em Never mind. Don't know why there was that error. Mine also works now.

closed 2017-06-10 by kylesf