udacity/robotics-beta-test-issues #477: vivek1108, l1-19

vxy10, 2017-06-01:
In link below, the forward kinematics calculations seems to be wrong, for q2 -> q2+pi/2. Even when all angles are substituted 0, the arm must be straight out, so z and x axis must match. it will be nice to clarify angles and zero configuration, if this is not the case. 

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/72800973-80cd-4828-81fd-b3390486892b/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/c0837700-3de6-4c41-8a5d-1e25936e0cdb

thomas-at-em, 2017-06-01:
Yes, you are correct. This is a copy-paste error. But you have given me a good idea as to how to expand this concept. Take a look at this lesson now and tell me what you think.

vxy10, 2017-06-01:
Nice, looks good. i like the quiz part too, gives more intuition

closed 2017-06-14 by sahiljuneja