udacity/robotics-beta-test-issues #478: vivek1108: DH parameters, common normal

vxy10, 2017-06-01:
Identify the common normal between frames {i-1} and frame {i}, it would be benefitial to write common normal between z-axes.

thomas-at-em, 2017-06-01:
I don't understand what problem you are identifying. Could you provide more details please?

vxy10, 2017-06-01:
In link below, item 4. It says identify common normal between frames i-1 and i. However, what we need to do is to identify common normal between z-axes of i-1 and i. Its a minor point, but will be nice to clarify. 

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/72800973-80cd-4828-81fd-b3390486892b/modules/8855de3f-2897-46c3-a805-628b5ecf045b/lessons/87c52cd9-09ba-4414-bc30-24ae18277d24/concepts/2f59c902-9c32-4b26-9e52-5e495ec14dba

thomas-at-em, 2017-06-02:
Hey, sorry for the late reply, I have been out sick. I see your point. Should be fixed now. Thanks!

closed 2017-06-02 by thomas-at-em