udacity/robotics-beta-test-issues #484: sjuneja P02 Pick and Place - Import Statements consistency for project vs lessons

sahiljuneja, 2017-06-04:
In the `IK_server.py` there is an important statement for `sympy`

```python
from sympy import *
```

For the `Forward Kinematics` lecture in the `Introduction to Kinematics` lesson, we have
```python
from sympy import symbols, cos, sin, pi,simplify
from sympy.matrices import Matrix
```

Since many students faced some issues with `numpy` already, and some might not be adept at python, do you think there needs to be some level of consistency to make it easier for them?

I would like to believe they will figure this out, but just wondering what you guys think. Let me know if it's too much nitpicking.

vxy10, 2017-06-04:
Another issue may be that sin in sympy has type of sin and sin(x) may return a symbolic object, but numpy returns a float.

Shreeyak, 2017-06-06:
I'm having quite some difficulty with figuring out how to correctly use sympy to solve IK in my notebook. Just a reference.

closed 2017-07-10 by sahiljuneja