udacity/robotics-beta-test-issues #495: Lesson 3.5(Arm Mover:The Code)

sychaichangkun, 2017-06-13:
In the code explanation of clamp_at_boundaries(), “~” is said to be the private namespace qualifier, representing '/simple_arm/' here. 
Anctually '~' is 'arm_mover' because the node name is 'arm_mover'.
You can imput 'rospy.get_param_names()' to see what '~' is.

sychaichangkun, 2017-06-13:
sorry '  ~  ' can not be displayed properly.

sychaichangkun, 2017-06-13:
In the code explanation of clamp_at_boundaries(), '  ~  ' is said to be the private namespace qualifier, representing '/simple_arm/' here.
Anctually '  ~   '   is  ' arm_mover ' because the node name is ' arm_mover '.
You can imput  ' rospy.get_param_names() ' to see what '  ~   ' is.

kylesf, 2017-07-28:
What is the status of this issue?

swwelch, 2017-08-04:
Updated. Thanks!

closed 2017-08-04 by swwelch