udacity/robotics-beta-test-issues #497: VM - 4K screen problems

sahiljuneja, 2017-06-13:
With a 4K screen, the VM has an issue where it shows a very small resolution/size corresponding to the text inside the VM.

Current solution is to manually increase font size. This seems to be a problem with Vmware I believe and might get fixed at some point. 

Might have to add the workaround (font size and any other) in the FAQ.

kylesf, 2017-06-13:
Can you share a screen shot of your Display options in the system settings?

sahiljuneja, 2017-06-13:
@kylesf Updated post.

sahiljuneja, 2017-06-13:
The font size change doesn't work for the terminal. Which is very annoying to look at right now 😅

kylesf, 2017-06-13:
Can you try to manually change the size of the monitor and then make the VM fullscreen?

As for text size, what terminal are you using?

sahiljuneja, 2017-06-13:
Yup, tried those. Didn't help. This is a known problem apparently. VMware Workstation Pro seems to have that fixed. 

The default one, LXTerminal. Can change the font size for that separately. But then trying to work with RViz and Gazebo and all, everything is like for ants! [Dunno if you get that reference]

I will mess around it with a bit and try for a workaround. But will probably move to virtual box or docker if nothing works. Won't personally be able to work with this.

kylesf, 2017-06-13:
Zoolander lol. You should be able to set the preferences for terminal and have it apply for all windows. I also recommend using terminator for ROS instead. Try playing with font there.

sahiljuneja, 2017-06-29:
No solution came out of the VMWare Workstation Player itself for me. 

Did manage with some changes to lubuntu itself. 

Steps to follow -

- In home directory, create file `.Xresources`
- In the file enter desired DPI value in this format: `Xft.dpi: 150` (change 150 to number that works best for you)
- Restart the VM

The above is the only thing that has worked reasonably without any noticeable performance issues. Although, even with a powerful laptop, the VM is a resource hog and still performs relatively poorly compared to a full installation. Which is expected I guess.

I will add this to the wiki section soon. Feel free to close it if nothing more to add.

closed 2017-07-19 by kylesf