udacity/robotics-beta-test-issues #498: crl VM 2.0.1

zenetio, 2017-06-13:
There is no rosrun installed and it is called in C++ component. I had to install

kylesf, 2017-06-13:
Can you explain the steps you had to take and where is failed to work? I see running the command `rosrun` does execute the correct command.

zenetio, 2017-06-13:
1. I was checking demo
2. At final of cycle, I could not see the bin in the next cycle
3. Checked the source code and saw a system call with rosrun
4. Tried to execute rosrun manually and got the message "rosrun not installed" and suggested to install rosbash
5. Installed rosbash and fixed demo issue

kylesf, 2017-06-13:
Thanks for sharing. Code base was recently updated. I will update classroom and add to future OTA VM update

closed 2017-06-29 by kylesf