udacity/robotics-beta-test-issues #500: crl L02E08

zenetio, 2017-06-15:
In the second paragraph I see "Consider the extrinsic matrix" but the equation is of intrinsic matrix

sahiljuneja, 2017-06-15:
Hey, the equation seems to be correct.

For Extrinsic, you pre-multiply the matrices. 

So  R_ZYX would have R_x*R_y*R_z

I think the confusion might possibly be because of a mistake in one of the frames in the video, which was just corrected. So if you can confirm on that, that would be great!

zenetio, 2017-06-16:
Maybe I am wrong but what I see in the text for extrinsic is
R_ZYX = R_zR_yR_x

sahiljuneja, 2017-06-16:
No, I think you are right. There is some inconsistency there. Thanks for pointing this out, I will check it out properly and fix it soon.

closed 2017-08-02 by sahiljuneja