udacity/robotics-beta-test-issues #502: Forward Kinematics Formula

jchernus, 2017-06-15:
Shows the same image twice. May I delete #2?

As well, the formula https://d17h27t6h515a5.cloudfront.net/topher/2017/June/593eecf9_eq2/eq2.png

is written in a different form from what we saw earlier,

https://d17h27t6h515a5.cloudfront.net/topher/2017/May/591e48cb_image-2/image-2.png

Can we keep it consistent?

thomas-at-em, 2017-06-15:
I would actually delete the first one. 
Keeping the equation consistent is a good idea. If you are not familiar with LaTeX, I think @sahiljuneja is.

sahiljuneja, 2017-06-15:
Yup, I will get to that soon.

jchernus, 2017-06-16:
I've been doing a lot of LaTeXing (and Ryan sent me your sheets) so it's an easy fix. Which formula do you prefer?

jchernus, 2017-06-16:
Also, being more compact, I think https://d17h27t6h515a5.cloudfront.net/topher/2017/June/59378aa2_screen-shot-2017-06-06-at-10.06.02-pm/screen-shot-2017-06-06-at-10.06.02-pm.png is easier to interpret than the other two images because you can see everything together without needing to scroll (maybe others are using bigger monitors than I and don't have this problem... )

thomas-at-em, 2017-06-16:
http://s2.quickmeme.com/img/62/628b747f8ccdfb757062f36a27eedecfc2295f515c0586e05fbfb0620c0571a2.jpg

I like the equation that appears in Forward Kinematics above the SCARA manipulator image:
https://d17h27t6h515a5.cloudfront.net/topher/2017/June/593eecf9_eq2/eq2.png

closed 2017-06-16 by jchernus