udacity/robotics-beta-test-issues #514: Lesson 2: Introduction to 3D Perception, Active vs Passive Sensors

cornzyblack, 2017-07-02:
I believe the first diagram for Active and Passive sensors is meant to give an idea of what Active and Passive sensors are. In the diagram, the visual representation for the Passive part is clear ->
sun rays hit a robot, and the video camera is able to measure the reflected rays without any need to release energy into the environment. However for the Active part the video camera sends energy into the environment, but nothing is reflected back. Just a thought, why not use a Camera with a flash for the Active sensor, and show the rays being reflected back, and a Camera without a flash for passive sensor

kylesf, 2017-07-05:
This is a great idea and we will keep it in mind for future updates!

ryan-keenan, 2017-07-05:
Just made this update!  Looks like this now
[image]

closed 2017-07-05 by ryan-keenan