udacity/robotics-beta-test-issues #515: Lesson 2: Introduction to 3D Perception, Active vs Passive Sensors - Comparison tables

thiagodefreitas, 2017-07-02:
First of all, I really like the comparison table, as it gives a better insight into what the student may face further on on the market, but sorry if I have missed something here, but IMHO, the fact that you have explicitly defined the interface for all sensor types as Ethernet only is a little bit misleading, as you may find those sensors with a broader range of interfaces(e.g SPI). Moreover, on the beginning of the text you mention 3D sensing techniques, although there seems to be 2D variants, such as the sonar description, what could also apply to LIDAR. Finally, LIDAR is then an specific case where the costs is a little bit tricky to define, where you could go from a Scanse Sweep, RPlidar to Velodyne or Sick that would cost more than 10 times more.

hvpandya, 2017-07-05:
Will add statements to make it clear that we are not referring to 2D planar lidars but only 3D lidars (which actually are high cost)

As far as Interfaces go, this table is a general comparison, there will always be outliers in every any set of data one compares. For comparing sensors across the isle, we sampled many variants of all these sensors and then generalized the results into the neat little table to provide a common ground for comparison.

hvpandya, 2017-07-05:
Fixed, thanks

closed 2017-07-05 by hvpandya