udacity/robotics-beta-test-issues #521: crl Perception, Lesson 3, E05

zenetio, 2017-07-03:
In the exercise, line 48, need replace dst by undist

https://classroom.udacity.com/nanodegrees/nd209-beta/parts/571766f0-9631-484d-a69f-94ea58110725/modules/e5bfcfbd-3f7d-43fe-8248-0c65d910345a/lessons/8d51e0bf-0fa1-49a7-bd45-e062c4a2121f/concepts/cd60c2cf-a017-45b0-a72e-25e9279e488f

zenetio, 2017-07-03:
The replace above is to remove the execution error. After correct the variable name, the execution is ok and I can get the images, but the submit is returning this message: 'module' object has no attribute 'undistorted'

zenetio, 2017-07-03:
In fact, to match with repo, the variable should be named dst

ryan-keenan, 2017-07-03:
Thanks for reporting this!  Fixed now

closed 2017-07-03 by ryan-keenan