udacity/robotics-beta-test-issues #524: arek - Project 2 - Running ./safe_spawner.sh

867sec, 2017-07-04:
When running `./safe_spawner.sh` in both demo `true` and demo `false` modes... getting the following error and cannot find any documentation about it... It happens every time in both VM 2.0.1 and 2.1.0 beta. Screenshot attached below. [ERROR] Found empty JointState message

sahiljuneja, 2017-07-04:
Since this is an isolated case, my best recommendation right now is that you start on the process again of downloading the repo and everything for the project and repeating the steps properly to ensure you checked off everything. 

Secondly, if this happens when you run the IK_server script, then it's quite possible you are not returning the correct values (or format) or there's some issue with your code somewhere, maybe? Share your script with the code with me on the course Slack, and I will try to see if I can reproduce the error on my end. 

But do try to redo the process without missing out on a step. Often there's something we end up missing out on which can cause these errors!

867sec, 2017-07-04:
This problem happens when running ./safe_spawner.sh not IK_server.py.
This problem happens on freshly downloaded VM version 2.0.1 or 2.1.0 after
following instructions in the "Setting up your environment" section of
lesson Project 2.
When you say: "repeating the steps properly to ensure you checked off
everything" please be specific. Do you mean the steps in the "Setting up
your environment" section of lesson Project 2 (which I did) or something
else?

hvpandya, 2017-07-22:
This is a benign issue related to a specific commit on moveit's kinetic-devel branch.

Since this is an isolated case and the errors do not affect the project in any manner other than being annoying, I am closing this issue.

More info: http://answers.ros.org/question/266274/moveit-regression-found-empty-jointstate-message/

closed 2017-07-22 by hvpandya